wled/WLED #4758: WLED interface changed background without update

jjx-, 2025-07-04:
### What happened?

Hi,
the background of my WLED interface changed from solid black from to a picture of a firework over the last night. I did no changes to the settings nor did I upgrade WLED. I am experiencing this issue on different ESPs running 0.14.2 onwards. On ESPs up to 0.14.0 I have no problems. Issue persists no matter if PC mode is enabled or not. Tested on Firefox 141.0b5

### To Reproduce Bug

Open WLED 0.14.2 or newer in Webbrowser.

### Expected Behavior

Black WLED background as always so you can find the buttons as easy as usual.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.2

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Adorfer, 2025-07-04:
looks like it's same issue on v0.15.0 release too on ESP32.

[image]

Adorfer, 2025-07-05:
> set background image opacity to 0
Would be helpful, to have at least a button for that. that solution was not obvious, at least not to me.