wled/WLED #4773: Extend the Usermod V2 Wordclock to 11x11 Pixels

Zippochonda, 2025-07-14:
I would like to add a 11x11 Layout from https://github.com/panbachi/wordclock/ to the 

**usermod_v2_word_clock**

The mask has to be changed, perhaps somebody could add a checkbox for this masking in the usermod?

` // defines for mask sizes
#define maskSizeLeds 121
#define maskSizeMinutes 12
#define maskSizeMinutesMea 12
#define maskSizeHours 6
#define maskSizeHoursMea 6
#define maskSizeItIs 5
#define maskSizeMinuteDots 4

// "minute" masks
// Normal wiring
const int maskMinutes[14][maskSizeMinutes] = 
{
  {107, 108, 109,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1}, // 0 - 00
  {  7,   8,   9,  10,  38,  39,  42,  43,  -1,  -1,  -1,  -1}, // 1 - 05 fünf nach
  { 11,  12,  13,  14,  40,  41,  42,  43,  -1,  -1,  -1,  -1}, // 2 - 10 zehn nach
  { 26,  27,  28,  29,  30,  31,  32,  -1,  -1,  -1,  -1,  -1}, // 3 - 15 viertel
  { 11,  12,  13,  14,  15,  16,  17,  40,  41,  42,  43,  -1}, // 4 - 20 zwanzig nach
  {  7,   8,   9,  10,  33,  34,  35,  44,  45,  46,  47,  -1}, // 5 - 25 fünf vor halb
  { 44,  45,  46,  47,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1}, // 6 - 30 halb
  {  7,   8,   9,  10,  40,  41,  42,  43,  44,  45,  46,  47}, // 7 - 35 fünf nach halb
  { 15,  16,  17,  18,  19,  20,  21,  33,  34,  35,  -1,  -1}, // 8 - 40 zwanzig vor
  { 22,  23,  24,  25,  26,  27,  28,  29,  30,  31,  32,  -1}, // 9 - 45 dreiviertel
  { 11,  12,  13,  14,  33,  34,  35,  -1,  -1,  -1,  -1,  -1}, // 10 - 50 zehn vor
  {  7,   8,   9,  10,  33,  34,  35,  -1,  -1,  -1,  -1,  -1}, // 11 - 55 fünf vor
  { 26,  27,  28,  29,  30,  31,  32,  40,  41,  42,  43,  -1}, // 12 - 15 alternative viertel nach
  { 26,  27,  28,  29,  30,  31,  32,  33,  34,  35,  -1,  -1}  // 13 - 45 alternative viertel vor
};

// Meander wiring
const int maskMinutesMea[14][maskSizeMinutesMea] = 
{
  {112, 114, 116,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1}, // 0 - 00
  {  7,   8,   9,  10,  38,  39,  40,  41,  -1,  -1,  -1,  -1}, // 1 - 05 fünf nach
  { 18,  19,  20,  21,  38,  39,  40,  41,  -1,  -1,  -1,  -1}, // 2 - 10 zehn nach
  { 26,  27,  28,  29,  30,  31,  32,  -1,  -1,  -1,  -1,  -1}, // 3 - 15 viertel
  { 11,  12,  13,  14,  15,  16,  17,  38,  39,  40,  41,  -1}, // 4 - 20 zwanzig nach
  {  7,   8,   9,  10,  35,  36,  37,  44,  45,  46,  47,  -1}, // 5 - 25 fünf vor halb
  { 44,  45,  46,  47,  -1,  -1,  -1,  -1,  -1,  -1,  -1,  -1}, // 6 - 30 halb
  {  7,   8,   9,  10,  38,  39,  40,  41,  44,  45,  46,  47}, // 7 - 35 fünf nach halb
  { 11,  12,  13,  14,  15,  16,  17,  35, 36, 37, -1,  -1}, // 8 - 40 zwanzig vor
  { 22,  23,  24,  25,  26,  27,  28,  29,  30,  31,  32,  -1}, // 9 - 45 dreiviertel
  { 18,  19,  20,  21,  35,  36,  37, -1,  -1,  -1,  -1,  -1}, // 10 - 50 zehn vor
  {  7,   8,   9,  10,  35,  36,  37, -1,  -1,  -1,  -1,  -1}, // 11 - 55 fünf vor
  { 26,  27,  28,  29,  30,  31,  32,  38,  39,  40,  41,  -1}, // 12 - 15 alternative viertel nach
  { 26,  27,  28,  29,  30,  31,  32,  35,  36,  37,  -1,  -1}  // 13 - 45 alternative viertel vor
};

// hour masks
// Normal wiring
const int maskHours[13][maskSizeHours] = 
{
  { 55,  56,  57,  -1,  -1,  -1}, // 01: ein
  { 55,  56,  57,  58,  -1,  -1}, // 01: eins
  { 62,  63,  64,  65,  -1,  -1}, // 02: zwei
  { 66,  67,  68,  69,  -1,  -1}, // 03: drei
  { 73,  74,  75,  76,  -1,  -1}, // 04: vier
  { 51,  52,  53,  54,  -1,  -1}, // 05: fünf
  { 77,  78,  79,  80,  81,  -1}, // 06: sechs
  { 88,  89,  90,  91,  92,  93}, // 07: sieben
  { 84,  85,  86,  87,  -1,  -1}, // 08: acht
  {102, 103, 104, 105,  -1,  -1}, // 09: neun
  { 99, 100, 101, 102,  -1,  -1}, // 10: zehn
  { 49,  50,  51,  -1,  -1,  -1}, // 11: elf
  { 94,  95,  96,  97,  98,  -1}  // 12: zwölf and 00: null
};
// Meander wiring
const int maskHoursMea[13][maskSizeHoursMea] = 
{
  { 61,  62,  63,  -1,  -1,  -1}, // 01: ein
  { 61,  62,  63,  64,  -1,  -1}, // 01: eins
  { 62,  63,  64,  65,  -1,  -1}, // 02: zwei
  { 67,  68,  69,  70,  -1,  -1}, // 03: drei
  { 77,  78,  79,  80,  -1,  -1}, // 04: vier
  { 73,  74,  75,  76,  -1,  -1}, // 05: fünf
  {104, 105, 106, 107, 108,  -1}, // 06: sechs
  { 55,  56,  57,  58,  59,  60}, // 07: sieben
  { 89,  90,  91,  92,  -1,  -1}, // 08: acht
  { 81,  82,  83,  84,  -1,  -1}, // 09: neun
  { 93,  94,  95,  96,  -1,  -1}, // 10: zehn
  { 85,  86,  87,  -1,  -1,  -1}, // 11: elf
  { 49,  50,  51,  52,  53,  -1}  // 12: zwölf and 00: null
};

// mask "it is"
const int maskItIs[maskSizeItIs] = {0, 1, 3, 4, 5};

// mask minute dots
const int maskMinuteDots[maskSizeMinuteDots] = {112, 114, 116, 118};

// overall mask to define which LEDs are on
int maskLedsOn[maskSizeLeds] = 
{
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0,
  0,0,0,0,0,0,0,0,0,0,0
};

[image]

PagedPenguin, 2025-07-14:
Sorry, I wrote the wrong PR number, Ignore the mention :3

Zippochonda, 2025-07-18:
Thanks, i added it to https://github.com/wled/WLED/pull/4781 

I hope thats the right way, i never made a Pull Request before.

closed 2025-07-18 by DedeHai