wled/WLED #4778: Flicker issues when sending real-time messages

jcelerier, 2025-07-17:
### What happened?

Hello!
I'm sending real-time controls for a device running https://github.com/intermittech/QuinLED-Firmware 
I'm having a flicker issue that I cannot find any satisfying answer for. I have 8 RGB 32-pixel led bars

With WLED's user interface lights control I don't see any flicker.

When I send ArtNet, sACN or DDP I get an occasional random flicker especially when I'm sending dark or even pure black: some random parts of random segments will vividly turn whiteI tried multiple rates of sending, from 20 to 120 hz to no avail.
But, again sending the same shade through WLED's internal interface work fine.

I tried with a very simple DDP sending code in C++ (attached here) that can reproduce said flickers.

What else can I try? what could be a cause of the flickers?

Here are wireshark recordings of the packets I am sending in artnet and ddp if anyone can see something wrong. I tried both with and without artnet sync, and in DDP all fits in one packet. 

https://github.com/user-attachments/assets/0f3f2b83-2d66-403f-8ffd-c69de486f137

https://github.com/user-attachments/assets/cc3593af-818e-4c2a-87c2-37dfe1d142e7

my ddp-sending code:

```
#include <algorithm>
#include <arpa/inet.h>
#include <cassert>
#include <chrono>
#include <cstring>
#include <netinet/in.h>
#include <stdio.h>
#include <sys/socket.h>
#include <unistd.h>

#pragma pack(push, 1)
struct ddp_packet
{
    //                                               v v x T S R Q P
    static constexpr auto DDP_PUSH_FLAGS = uint8_t{0b0'1'0'0'0'0'0'1};
    uint8_t flags = DDP_PUSH_FLAGS;

    // to increment
    //                 x x x x n n n n
    uint8_t sequence{0b0'0'0'0'0'0'0'0};

    //                                              C R TTT SSS
    static constexpr auto DDP_RGB8_TYPE = uint8_t{0b0'0'001'011};
    uint8_t data_type = DDP_RGB8_TYPE;

    // ID
    static constexpr int DDP_ID_DISPLAY = 1;
    static constexpr int DDP_ID_CONFIG = 250;
    static constexpr int DDP_ID_STATUS = 251;
    uint8_t src_dest_id = DDP_ID_DISPLAY;

    uint32_t data_offset{};

    uint16_t data_length{};

    static constexpr int DDP_MAX_DATALEN = (480 * 3);
    uint8_t data[4 + DDP_MAX_DATALEN];
};
#pragma pack(pop)

int main()
{
    using namespace std::literals;
    int sockfd = socket(AF_INET, SOCK_DGRAM, 0);
    if (sockfd < 0) {
        perror("Failed to create socket");
        return 1;
    }

    sockaddr_in dest_addr;
    memset(&dest_addr, 0, sizeof(dest_addr));
    dest_addr.sin_family = AF_INET;
    dest_addr.sin_port = htons(4048);
    inet_pton(AF_INET, "192.168.0.190", &dest_addr.sin_addr);

    ddp_packet packet{};

    packet.data_length = htons(8 * 32 * 4);
    packet.data_offset = 0;
    std::fill_n(packet.data, sizeof(packet.data), 0);

    static constexpr auto frame_duration = 33ms;
    auto next_frame = std::chrono::steady_clock::now();

    while (true) {
        ssize_t sent = sendto(sockfd, &packet, sizeof(packet), 0,
                             (struct sockaddr*)&dest_addr, sizeof(dest_addr));
        assert(sent == sizeof(packet));

        packet.sequence = (packet.sequence + 1) & 0b00001111;
        next_frame = std::chrono::steady_clock::now() + frame_duration;

        // Busy wait to maximize precision to make sure this is not a problem
        // due to incoming packet jitter. With wireshark I measure on average
        // less than 5 microseconds of jitter from the 30hz frequency.
        while (std::chrono::steady_clock::now() < (next_frame - 2500ns))
            ;
    }

    close(sockfd);
    return 0;
}
```

### To Reproduce Bug

Send low light intensity over DDP, ArtNet or s/ACN

### Expected Behavior

No flickers

### Install Method

Self-Compiled

### What version of WLED?

0.15

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jcelerier, 2025-07-19:
Hmm, in the post #4389 the flickers happen with the built-in presets of WLED.
In my case it only happens when sending external messages through Art-Net / sACN / DDP, but the built-in presets and controls never exhibit the flickers so I'm not sure it could be a hardware issue :/

DedeHai, 2025-07-19:
@jcelerier it is a very complex issue and has to do with hardware timing and buffer handling. The higher the load, the more likely it is to happen.

jcelerier, 2025-07-19:
I see. In my case the flickers happened starting from 20hz (I didnt't see any at lower frequencies but maybe it's just luck) and did not seem to be meaningfully different from when I was sending at 120hz. 

I also tried running the presets and just sending an ICMP ping at high rate (1khz) to see if this would cause disturbance but did not see any flickers there when using the builtin presets. But maybe the ethernet chip is processing ping without forwarding that to the ESP so idk.