wled/WLED #4782: Guru Meditation Error: Core  1 panic'ed if pin not available

maxx-ukoo, 2025-07-18:
### What happened?

I have board with ethernet. Ethernet chip use pins 16/17 for communication.
It looks like by default WLED use pin 16 as data pin for W28xx strips.
As result W28xx object not initialized properly and app crash on attempt of control leds.

### To Reproduce Bug

Flash ethernet firmware on ethernet supported board with pin 16/17 as data pins and configure ethernet.

### Expected Behavior

If data pin not available WLED shouldn't run effects on led object

### Install Method

Self-Compiled

### What version of WLED?

2412040 v4 eth build from https://github.com/wled/WLED/compare/0_15_x...maxx-ukoo:WLED:mk_0_15_0_RTL8201?expand=1 branch

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

netmindz, 2025-07-27:
We should not be panic like this, there is code to check if the pin is available and we have checks like  ` if (!_valid) return;` so why are we still seeing the crash? Do you have the full crash log @maxx-ukoo ?

netmindz, 2025-07-27:
Please can you check if my commit was enough to resolve the issue you were having @maxx-ukoo ?

qistoph, 2025-08-26:
https://github.com/wled/WLED/commit/e5ba97bbe2e077438b5a93ee16406b81f4a8ab75 introduces issues for my `PWM RGB`

I am running a "Wemos D1 mini" with RGB outputs on GPIO `5`, `4`, and `3`. Updating my firmware with `main`/`nightly` today the output didn't work anymore. Using bisect I found this commit to be the issue.

```
git bisect start
# status: waiting for both good and bad commits
# bad: [1fb9eb771e2cb6f9163f2ce695acb3c1bedd8249] Merge pull request #4856 from wled/copilot/fix-4855
git bisect bad 1fb9eb771e2cb6f9163f2ce695acb3c1bedd8249
# status: waiting for good commit(s), bad commit known
# good: [4f3992b4dc57c888b53d9576c894a278e03e4cfc] 0.15.1
git bisect good 4f3992b4dc57c888b53d9576c894a278e03e4cfc
# good: [a121f5b61b2dbb2934630120f22527a5cdab91d4] Merge pull request #4286 from blazoncek/xml-bugfix
git bisect good a121f5b61b2dbb2934630120f22527a5cdab91d4
# good: [2eff6b7a3a202638ac96b33121cea8b5ee3b1fd9] usermod/sensors_to_mqtt: Add explicit dep
git bisect good 2eff6b7a3a202638ac96b33121cea8b5ee3b1fd9
# good: [849d5e6667c2f5e682281d21f46dd995606e006c] Add libArchive to other usermods
git bisect good 849d5e6667c2f5e682281d21f46dd995606e006c
# good: [9099b13f692275978c8848923f29dd536fa5938a] Fix deps for Si7021_MQTT_HA
git bisect good 9099b13f692275978c8848923f29dd536fa5938a
# good: [8bcd4550f898129f07157b321744dcb73db3045b] Merge pull request #4762 from willmmiles/parallel-i2s-bus-selection
git bisect good 8bcd4550f898129f07157b321744dcb73db3045b
# bad: [93e011d4039d1a1e8c4a70503f2d79cf4ac44c3b] Merge pull request #4769 from wled/fix-4671
git bisect bad 93e011d4039d1a1e8c4a70503f2d79cf4ac44c3b
# bad: [591dbe387cb67de0621c6fcbea4f4a2b78aaa55c] Merge pull request #4780 from jasonsomers/main
git bisect bad 591dbe387cb67de0621c6fcbea4f4a2b78aaa55c
# good: [ab5b6f9b7dd36f4ec865caaea5eb896df12d4a21] fix change that got lost
git bisect good ab5b6f9b7dd36f4ec865caaea5eb896df12d4a21
# good: [806163f1edeed1133871a8959d204e5d6bac3a54] Merge branch 'wled:main' into main
git bisect good 806163f1edeed1133871a8959d204e5d6bac3a54
# bad: [e5ba97bbe2e077438b5a93ee16406b81f4a8ab75] Prevent crash during blend if led init fails, https://github.com/wled/WLED/issues/4782
git bisect bad e5ba97bbe2e077438b5a93ee16406b81f4a8ab75
# good: [b79e81f3be5dd0d75cae3f163bebdb19493feb89] maxIssues for release notes
git bisect good b79e81f3be5dd0d75cae3f163bebdb19493feb89
# first bad commit: [e5ba97bbe2e077438b5a93ee16406b81f4a8ab75] Prevent crash during blend if led init fails, https://github.com/wled/WLED/issues/4782
```

blazoncek, 2025-08-26:
@netmindz your fix from https://github.com/wled/WLED/commit/e5ba97bbe2e077438b5a93ee16406b81f4a8ab75 seems incorrect and does not really resolve the underlying issue.

A better way is to change default GPIO assignment during compilation by changing this line: https://github.com/wled/WLED/blob/1fb9eb771e2cb6f9163f2ce695acb3c1bedd8249/wled00/const.h#L577

to

```c++
#if defined(ESP8266) || defined(CONFIG_IDF_TARGET_ESP32C3) || defined(WLED_USE_ETHERNET)
```

It may still cause issues if DMX is also compiled (as GPIO1 may also be used by some Ethernet boards), but at least it is not a default build.

softhack007, 2025-12-24:
Same issue as described here
* https://github.com/wled/WLED/issues/5155#issuecomment-3614391561

It looks like the safest solution - as we should avoid any pin used by at least one ethernet config - will be to use gpio 4 in ethernet enabled builds.