wled/WLED #4788: 2D "Matrix Generator" issue

DedeHai, 2025-07-24:
### What happened?

When using the 2D matrix generator, there are some minor issues (web UI only):
- When generating panel layouts, the layout gets generated but the preview is not updated properly until a value in the list is changed
- the setting in the generator for first pixel location and serpentine, it is ignored in the generated list

### To Reproduce Bug

set up a grid of panels, 2x2 for example using the matrix generator

### Expected Behavior

web UI values shoudl all be generated from input values, preview should be updated when clicking generate.

### Install Method

Self-Compiled

### What version of WLED?

0.16

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

not a pressing issue, just something I noticed.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-07-25:
> * the setting in the generator for first pixel location and serpentine, it is ignored in the generated list

Generator creates panel order in this configuration, not individual panel configuration.

DedeHai, 2025-07-25:
ah, that is solved then :)
leaves the missing update call for the preview.

closed 2025-09-03 by DedeHai