wled/WLED #4800: ESP crashes when changing from 1D stip to 2D Matrix and clicking on save

Andro-Marian, 2025-07-31:
### What happened?

Crashes when changing from 1D to 2D and click on save

### To Reproduce Bug

1. Fresh install the latest beta 0.16.
2. Open settings -> 2D Configuration -> Strip or panel: change to 2D Matrix -> Click: Save

```
The connection was reset

The connection to the server was reset while the page was loading.

    The site could be temporarily unavailable or too busy. Try again in a few moments.
    If you are unable to load any pages, check your computer’s network connection.
    If your computer or network is protected by a firewall or proxy, make sure that Firefox is permitted to access the web.```

### Expected Behavior

To save the settings

### Install Method

Binary from WLED.me

### What version of WLED?

0.16

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Backtrace: 0x4008ab44:0x3ffe11f0 0x4008aea1:0x3ffe1210 0x401c0557:0x3ffe1230 0x401680e3:0x3ffe1260 0x4013c1b4:0x3ffe1280 0x40118ba5:0x3ffe12a0 0x4013c7c6:0x3ffe1320 0x4013c827:0x3ffe13b0 0x4008c92a:0x3ffe13e0
  #0  0x4008ab44:0x3ffe11f0 in invoke_abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #1  0x4008aea1:0x3ffe1210 in vApplicationStackOverflowHook at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648    
  #2  0x401c0557:0x3ffe1230 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<WiFiClientRxBuffer*>(WiFiClientRxBuffer*) at C:\Users\DesktopTV\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiClient.cpp:572
      (inlined by) std::__shared_ptr<WiFiClientRxBuffer, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<WiFiClientRxBuffer>(WiFiClientRxBuffer*) at c:\users\desktoptv\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:882
      (inlined by) void std::__shared_ptr<WiFiClientRxBuffer, (__gnu_cxx::_Lock_policy)2>::reset<WiFiClientRxBuffer>(WiFiClientRxBuffer*) at c:\users\desktoptv\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:1030
  #3  0x401680e3:0x3ffe1260 in tcp_trigger_input_pcb_close at ??:?
  #4  0x4013c1b4:0x3ffe1280 in DNSServer::replyWithIP() at C:\Users\DesktopTV\.platformio\packages\framework-arduinoespressif32\libraries\DNSServer\src/DNSServer.cpp:189
  #5  0x40118ba5:0x3ffe12a0 in EspalexaDevice::getRGB() at wled00/src/dependencies/espalexa/EspalexaDevice.cpp:312
  #6  0x4013c7c6:0x3ffe1320 in AsyncUDP::listenMulticast(ip_addr const*, unsigned short, unsigned char, tcpip_adapter_if_t) at C:\Users\DesktopTV\.platformio\packages\framework-arduinoespressif32\libraries\AsyncUDP\src/AsyncUDP.cpp:411
  #7  0x4013c827:0x3ffe13b0 in AsyncUDP::listenMulticast(ip_addr const*, unsigned short, unsigned char, tcpip_adapter_if_t) at C:\Users\DesktopTV\.platformio\packages\framework-arduinoespressif32\libraries\AsyncUDP\src/AsyncUDP.cpp:411
  #8  0x4008c92a:0x3ffe13e0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Andro-Marian, 2025-07-31:
Or:
```
Backtrace: 0x4008ab44:0x3ffddea0 0x4008aea1:0x3ffddec0 0x40090209:0x3ffddee0 0x400862f6:0x3ffddf00 0x4008678d:0x3ffddf20 0x4000bec7:0x3ffddf40 0x4013d2bb:0x3ffddf60 0x4013d2c9:0x3ffddf80 0x40122af6:0x3ffddfa0 0x40122d81:0x3ffde080 0x40137a19:0x3ffde0a0 0x401c53ed:0x3ffde0f0 0x4013818d:0x3ffde110 0x40133dcd:0x3ffde150 0x4013537a:0x3ffde190 0x401353a9:0x3ffde1c0 0x4012a8a1:0x3ffde1e0 0x4012ad22:0x3ffde220 0x4012ad9b:0x3ffde240 0x4008c92a:0x3ffde270
  #0  0x4008ab44:0x3ffddea0 in invoke_abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #1  0x4008aea1:0x3ffddec0 in abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #2  0x40090209:0x3ffddee0 in multi_heap_free at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:305    
  #3  0x400862f6:0x3ffddf00 in heap_caps_free at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #4  0x4008678d:0x3ffddf20 in _free_r at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #5  0x4000bec7:0x3ffddf40 in ?? ??:0
  #6  0x4013d2bb:0x3ffddf60 in String::invalidate() at C:\Users\DesktopTV\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:859
  #7  0x4013d2c9:0x3ffddf80 in String::~String() at C:\Users\DesktopTV\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:859
  #8  0x40122af6:0x3ffddfa0 in serveSettings(AsyncWebServerRequest*, bool) at wled00/wled_server.cpp:669
  #9  0x40122d81:0x3ffde080 in std::_Function_handler<void (AsyncWebServerRequest*), initServer()::{lambda(AsyncWebServerRequest*)#10}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at wled00/wled_server.cpp:300
      (inlined by) _M_invoke at c:\users\desktoptv\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871     
  #10 0x40137a19:0x3ffde0a0 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at .pio\libdeps\esp32dev\ESPAsyncWebServerWLED\src/DynamicBuffer.h:259
      (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at .pio\libdeps\esp32dev\ESPAsyncWebServerWLED\src/WebHandlerImpl.h:132
  #11 0x401c53ed:0x3ffde0f0 in AsyncWebServerRequest::_handleRequest() at .pio\libdeps\esp32dev\ESPAsyncWebServerWLED\src/WebRequest.cpp:1048     
  #12 0x4013818d:0x3ffde110 in AsyncWebServer::processQueue() at .pio\libdeps\esp32dev\ESPAsyncWebServerWLED\src/DynamicBuffer.h:259
  #13 0x40133dcd:0x3ffde150 in AsyncWebServerRequest::_requestReady() at .pio\libdeps\esp32dev\ESPAsyncWebServerWLED\src/WebRequest.cpp:1048      
  #14 0x4013537a:0x3ffde190 in AsyncWebServerRequest::_onData(void*, unsigned int) at .pio\libdeps\esp32dev\ESPAsyncWebServerWLED\src/WebRequest.cpp:1048
  #15 0x401353a9:0x3ffde1c0 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#5}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at .pio\libdeps\esp32dev\ESPAsyncWebServerWLED\src/WebRequest.cpp:1048
      (inlined by) _M_invoke at c:\users\desktoptv\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871     
  #16 0x4012a8a1:0x3ffde1e0 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at .pio\libdeps\esp32dev\AsyncTCP\src/AsyncTCP.cpp:1123
      (inlined by) AsyncClient::_recv(pbuf*, signed char) at .pio\libdeps\esp32dev\AsyncTCP\src/AsyncTCP.cpp:947
  #17 0x4012ad22:0x3ffde220 in AsyncClient_detail::handle_async_event(AsyncClient_event_t*) at .pio\libdeps\esp32dev\AsyncTCP\src/AsyncTCP.cpp:1123
  #18 0x4012ad9b:0x3ffde240 in _async_service_task(void*) at .pio\libdeps\esp32dev\AsyncTCP\src/AsyncTCP.cpp:1123
  #19 0x4008c92a:0x3ffde270 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

blazoncek, 2025-07-31:
Crash dumps point to issues with networking stack. As with your previous report I am suspecting not WLED related.
You may also want to monitor your network traffic if there are any unusual packets broadcast.

Andro-Marian, 2025-07-31:
I don't think is the internet. Is different. And just when saving on the 2D setting page. No problem on 0.15 release for some reason.
And again. no problem on other versions.

Andro-Marian, 2025-07-31:
Crashes even in AP.

I pointing to: serveSettings(AsyncWebServerRequest*, bool) at wled00/wled_server.cpp:669

closed 2025-07-31 by Andro-Marian

Andro-Marian, 2025-08-03:
0.15.1 seams to work fine on this

Andro-Marian, 2025-08-04:
I think the: https://github.com/wled/WLED/pull/4658 add this bug somehow

[image]

DedeHai, 2025-09-17:
There are still bugs in 2D setup, but I was not able to find it yet. see https://github.com/wled/WLED/issues/4911

Gawd21, 2025-09-17:
Okay, this fluke is crazier that I first thought. I just tried to replicate the crashes on two sister boards and I tried in 2412040 and 2509170. Lights on/off, 2, 100, 512, 1024 LEDs, 1 8x8, 1 16x16, 2 16x16, 4 16x16, one board has audioreactive enabled the other does not. While using my phone's OTG to power the boards, and neither will crash now. They both even updated with 2d enabled. I'll test using a different PS, as all the tests I've ran today I was using OTG, and I didn't do that last night.