wled/WLED #4805: Seit FW 15.01 blinkt ein LED Strip

hardl1, 2025-08-02:
### What happened?

An WLED sind 2 LED-Strips WS2812B mit 300 LED`s an GPO 16 + 02 angeschlossen und funktioniert problemlos.

### To Reproduce Bug

Mit FW 15.0 ok, seit 15.1 blinkt ein Streifen im Abstand von 10 - 30 Sekunden.

### Expected Behavior

Kein Blinken, nur Unilight.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.1 (Build 2507300)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-08-02:
English, please.

hardl1, 2025-08-02:
> blazoncek
>  left a comment
> (wled/WLED#4805)
> English, please.

Sorry, chanced.

hardl1, 2025-08-02:
Unfortunately I can't find this setting.

BerendWouters, 2025-08-02:
I've got the same issue. I've got 4 segments, and (I think) two of them are flickering.

hardl1, 2025-08-02:
@BerendWouters,  not by me...

[image]

BerendWouters, 2025-08-02:
> @BerendWouters,  not by me...
> 
> [image]
It's introduced in 0.15.1, I think :)

hardl1, 2025-08-02:
That's right, I updated again to 0.15.1, the field is there, it's enabled, but it still flashes.

hardl1, 2025-08-02:
The entire strip is flashing. I can't comment on the level shifter, as I bought the device ready-made.

willmmiles, 2025-08-02:
Could anyone experiencing this issue please post their cfg.json file?  I haven't been able to replicate it locally, but I have a suspicion it might be related to exact LED strip lengths.

hardl1, 2025-08-02:
How do I do that? 
I'm back on 0.15.0, but I can change that.

willmmiles, 2025-08-02:
> How do I do that? I'm back on 0.15.0, but I can change that.

Settings..Config..Security and Updates; push 'Backup Configuration' and it will download the cfg.json file.   You don't need to change software versions; how the software handles the cfg from the older version is relevant to testing.

I've got one from another channel I'm testing now.

therealhalifax, 2025-08-02:
I have similar issue with 15.1 (downgrade to 15.0 resolves it) I see flashing when changing brightness of the wled via home assistant. In any case and for all my strips. And I see flashing when open the wled app on my phone or bring the app back in the front (after using another app) in all this cases Tha strips are flashing (switching off for some ms and on again)

BerendWouters, 2025-08-03:
Attached you'll find my config. I'm unsure if the lengths of the strip match the configured values.
[wled_cfg.json](https://github.com/user-attachments/files/21565986/wled_cfg.json)

bkeifer, 2025-08-03:
I'm seeing this as well, but only with the first 60-ish LEDs of one strip (labelled Southeast in the config).  Config attached.

[wled_cfg.json](https://github.com/user-attachments/files/21568193/wled_cfg.json)

Plawasan, 2025-08-05:
Same problem on both controllers I updated so far, both are Gledopto ESP32 based

[wled_cfg-LR-Shelves.json](https://github.com/user-attachments/files/21605718/wled_cfg-LR-Shelves.json)

[wled_cfg-TV-Board.json](https://github.com/user-attachments/files/21605773/wled_cfg-TV-Board.json)

myhrmans, 2025-08-08:
Same issue here on QuinLed Uno after upgrading from 14.0 - 15.1, and also trying 16.0 dev.

ensonic, 2025-08-10:
Same issue here. ESP32 with 9 pins driving 100 leds each. Was working fines for 2 years before. Already disabled all the I2S pins as they were conflicting with the led-pins.

distante, 2025-08-10:
Here an extra config: [wled_cfg_wled-dana-cama.json](https://github.com/user-attachments/files/21706380/wled_cfg_wled-dana-cama.json)

Also, at least on my devices, I can always reproduce the problem going from or to the "waterfall" effect to almost any other, specially "solid"

gmow1, 2025-08-12:
Same Issue

ESP32
with level shifter
2 legs, leg A fine, leg B starts flashing/issues at led 79(approx) and the rest of the leg.
working fine until 0.15.1
reverted back to 0.15.0 and all fine

distante, 2025-08-17:
@Fauch123456789 you can backup your presets and configuration in the settings in case something fails. Then flash the 0.15.0 version. I did that and I did not lose anything.

kopierschnitte, 2025-08-17:
@Fauch123456789 I even downgraded without the need to restore anything. Took 3min and everything was fine again.

Fauch123456789, 2025-08-17:
> [@Fauch123456789](https://github.com/Fauch123456789) I even downgraded without the need to restore anything. Took 3min and everything was fine again.

Thank you both, I downgraded back to 0.15.0 and the problem is gone so problem solved!  I did not need to reload my presets.

Tykwondo1, 2025-08-27:
same issue here

netmindz, 2025-08-28:
Can I please ask that anyone commenting, that they include details of their led config, e.g how many outputs, what led type, number of pixels

marmurek, 2025-10-31:
Hello guys i've encountered same issue. Tried updating with beta versions and the bug appears in 0.15.1-beta.2. Previous - 0.15.1-beta.1 works OK.

My hardware is: Official SP511e without modifications, 5V 5,5Amps power supply, single output - 100 leds (10m) of the string lights with WS281x.

Dunno if this is any useful information, but i have quite big voltage drop at the end of lights, goes down to 3,3v, but i tried plugging another 5V 2,5amps power supply at the end and it didnt change anything (Apart from the fact that it works ok on the older version)

Led config as below:

[image]

Regarding the changelog, the only one change i found between these two version is #4484

marmurek, 2025-11-03:
Ok, i've managed to compile release 0.15.1 with reverted makuna/NeoPixelBus @ 2.8.0 and the problem is gone. I think this confirms my previous suspicions that refactoring done in NeoPixelBus 2.8.1 broke something.

I've attached own-built binary for ESP8266_2m, maybe someone else can confirm this as well.

[WLED_0.15.1_ESP02.bin.gz](https://github.com/user-attachments/files/23300203/WLED_0.15.1_ESP02.bin.gz)

hardl1, 2025-11-03:
Too bad, I can't test it because I'm using an ESP32.

bkeifer, 2025-12-02:
Unfortunately, I'm still seeing the behavior with multiple QuinLed DigQuads.

intermittech, 2025-12-02:
> Unfortunately, I'm still seeing the behavior with multiple QuinLed DigQuads.

Can you explain in your own words what you are seeing? Or maybe a photo or small video? Just to make sure it's the issue of this change. And I guess most important, if you don't see the same behaviour using v0.14.x.

hardl1, 2025-12-03:
For me too, version 0.15.2 is worse than version 0.15.1.
One strip is a solid turquoise, the other flashes constantly in all colors.

Back to 0.14.4 it's almost ok (brief flashing every few minutes)

distante, 2025-12-03:
My esp32 wroom boards are now working correctly with  0.15.2. I also sended the device logs from the new pop-up.

hardl1, 2025-12-23:
@ DedeHai Thank you for your willingness to help. I won't be home over the holidays to check the cables. As far as I remember, the cable from the WLED to the first LED is about 1 meter long.

closed 2026-08-30 by github-actions[bot]