wled/WLED #4806: gledopto gl lb 003 won't update to wled 0.15.1

Shiroe93, 2025-08-03:
### What happened?

i'm not able to update the gl lb 003 when i try i with WLED_0.15.1_ESP32.bin the update fails and tells me to check the file (i checked on gledopto reddit and i'm sure that's the right version to use)
any suggestions? 
the previous update from 0.13 to 0.15 completed without issue

### To Reproduce Bug

load the ota update in the ui 
press update 
(i can't flash the firmware directly because i don't see the device when connected to a pc)

### Expected Behavior

update is successful 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (build 2412100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Shiroe93, 2025-08-03:
This one right?
[image]

Arcitec, 2025-08-17:
> (i can't flash the firmware directly because i don't see the device when connected to a pc)

Does the device have a USB port?

If so, Gledopto's older controllers unfortunately suffered from a `5V -> 3.3V` voltage converter that sometimes isn't able to get enough voltage from computer USB ports. Basically the converter's threshold is too high.

You can "fix" that by getting an external 5 volt power adapter and plugging that in at the same time as the USB cable. It's very important that you use a 5 volt adapter! Higher voltages don't mix well with 5V USB and may damage your USB port!

I hope this helps you. Flashing via USB is definitely the cleanest way to get a guaranteed flash. Just be sure to follow the guides and flash the `esp32_bootloader_v4.bin` bootloader to 0x0 and the firmware to 0x10000.

Arcitec, 2025-08-17:
Ohhh it's a light bar. Yeah I don't think that product has the CH340 chip to be able to program via USB. I looked at the product page and it really just seems like it uses a USB power plug, not a data plug.

So you'll have to use WiFi OTA updates. Which I have no idea why it's not working for you, since 0.15 and 0.15.1 use the same partition layout. It should "just work".

Shiroe93, 2025-09-02:
> Check your WLED settings to be sure OTA flashing is not locked in WLED's settings etc

This are my settings let me know if you see anything wrong please 
[image]