wled/WLED #4807: 0.16 Gif Preset Bug

BerryMods, 2025-08-03:
### What happened?

When I click a preset for the gif it plays perfectly, but when i click a normal effect after playing a gif preset the whole matrix just goes blank.

If I go to `Settings > 2D Configuration` and just click save without even changing anything it lets me use the normal effects again but as soon as I start a Preset the normal effects stop working again

### To Reproduce Bug

1. I installed 0.16 with this installer: https://wled-install.github.io/
2. I set the LED count and set it to a matrix in settings
3. I went to the Pixel Magic Tool, set the settings, uploaded the gif, then started the preset for it
4. After all this the normal effects stopped working until I went to 2D Configuration settings and saved again without even changing anything

### Expected Behavior

I expected the normal effects to still work

### Install Method

Binary from WLED.me

### What version of WLED?

0.16

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-08-03:
Did you click on eye icon?

DedeHai, 2025-09-30:
@superagentorange this issue is about the pixel magic tool, not the gif player. I checked what you wrote and cannot reproduce, the gif player works fine with other effects.

superagentorange, 2025-10-01:
Ok sorry i thought the issue might be related to gif (image) playback and "Pixel Magic Gif" would be the same. Thank you for trying and strange that it does not behave the same for you. Maybe I open another issue for this problem?

I just tried the nightly from today and have the same behaviour.

After I have run one effect (does not matter which) the LEDs stay black for any GIF i choose (preset, through gifplayer.htm or by uploading a new GIF). It also shows the same behaviour (black screen) on the "peek" preview.

closed 2026-02-07 by github-actions[bot]