wled/WLED #4820: Add power limiting for PWM LEDs

Taudris, 2025-08-06:
**Is your feature request related to a problem? Please describe.**
I have more LED strip than my power supply is rated to handle, so I want to limit the max brightness to reduce the load on the PSU. The options available for digital LEDs are extensive, but for PWM LED outputs, there is no way to specify the current required for each output.

Digital LED output options:
[image]

PWM LED output options:
[image]

**Describe the solution you'd like**
1. Automatic brightness limiting should be able to be used with PWM LED outputs.
1. Each GPIO should have a configurable current. The CCT LED strips I'm using consume about the same amount of power for each color; if that's generally true for multichannel LED strips, it would be convenient to have a "current per channel" setting that applies to all channels of the LED output.

**Describe alternatives you've considered**
I mean, I guess I could buy a PSU with a higher power rating, but why would I do that when a software solution is possible? 😁

Taudris, 2025-08-06:
I see a setting called brightness *factor,* is that what you’re thinking of? It seems to do exactly what I need. Darn, wish I’d found it before spending time making this issue. I tried to find something like it originally, I swear!

I think it’s still a reasonable ask overall, but since I don’t personally need it now, should I close this issue?

blazoncek, 2025-08-11:
Just FYI: *Brightness factor* is not intended for current limitation. It does no checking and only multiplies brightness value (by 0.01 - 2.55).

Taudris, 2025-08-11:
Yes, I understand. I just figured I can measure the peak current consumption of the LED strip on a beefier power supply, then calculate a % with some derating to use as the brightness factor for the weaker supply. Better than nothing.

I also went and bought a power supply with a higher power rating anyway because this "24V" unit only outputs 21.6V. Sucks for mounting and wiring by comparison, but oh well.