wled/WLED #4839: WS2805 ArtNET/sACN

adhdale, 2025-08-14:
I am trying to get my WS2805 strip/pixels to work with artnet/sacn. I can set it up and it receives the signal, but I cannot properly control the RGBCW channels. If I put red on full, I see blank, red, green, blue, white repeating. There is no way I can get the colors working properly. 

Would it be possible to have the artnet/sacn have a profile that drives the RGBCW properly. A Multi RGBWW profile?

I have tried everything I know

Please let me know if I can submit more info.

netmindz, 2025-08-15:
There is no support for direct control of each colour channel you list there and it's not possible due to the way WLED works internally, you can only use the automatic estimation of the extra channels

closed 2025-08-15 by netmindz

adhdale, 2025-08-15:
How would I go about that, I don't mind not have CCT but I would like to do anything properly with the strip. I just get weird patterns as I mentioned, nothing as it should be shown.