wled/WLED #4850: WLED AP not showing after Factory Reset

tomware03, 2025-08-19:
### What happened?

I bought a WLED controller of amazon from a brand called GLEDOPTO. I have been having issues connecting it to with alexa so I decided a factory reset might help. After factory reseting, I cannot connect to the device at all anymore. Can you help fix this please?

### To Reproduce Bug

Factory resset device

### Expected Behavior

Reload and show AP on wifi settings

### Install Method

Binary from WLED.me

### What version of WLED?

0.16.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-08-19:
Raise this issue with the manufacturer and please use Discord or discourse forum for support questions.

willmmiles, 2025-08-19:
Possible duplicate of #4868.

closed 2025-08-20 by DedeHai