wled/WLED #4862: ability to choose presets based on PWM (servo type signal) input to ESP32

arkarkark, 2025-08-22:
**Is your feature request related to a problem? Please describe.**

I use wled on my little model airplanes controlled by ELRS and it would be super neat to be able to choose a preset from my radio. The receiver on my plane can output PWM signals in the range 1,000 µsec to 2,000 µsec with 1,500 µsec being the center value. It can also output SBUS and CRSF protocols, but PWM would be simplest to use. Also you could use a servo tester to test it during development.

**Describe the solution you'd like**

The ability to specify which pin would detect a PWM value and then what to do with that value. Most simple would be to map the full servo range to a range of presets. But I could imagine it being more complicated too where you specify signal timing ranges and which preset they apply to. or even controlling parameters of effects with values from a PWM port.

I'm not 100% sure that the voltage of the signals that servo testers or receivers output would be in the range that an ESP32 would accept on an input port so maybe a voltage divider would be needed.

blazoncek, 2025-08-22:
You can use analog "button" for such purpose. Not PWM though.

netmindz, 2025-09-03:
There has also been some discussion of SBUS as well I believe, possibly an issue on GitHub