wled/WLED #4863: Support for multiple apa102 strips stopped working correctly in v0.15.x+

kylix-rd, 2025-08-22:
### What happened?

Cannot add a 3rd strip of apa102 LEDs (option is disabled).
Only the first apa102 strip works.

4-apa102 strips w/ 140 LEDs each

data/clock pins
strip 0 - 15 4
strip 1 - 16 17
strip 2 - 5 18
strip 3 - 19 21

Each above line buffered through a 74HCT245 with series 100ohm resistor.

### To Reproduce Bug

Attach 3+ apa102 strips to ESP32 with separate clock/data pins for each strip. Each strip contains 140 LEDs
In LED settings page, you can add at most 2 apa102 strips and set the clock/data pins. Only the first strip works and lights up.
Trying to add a 3rd apa102 strip fails because the apa102 option is disabled.

### Expected Behavior

Up to v0.14.4, able to add 4 apa102 strips each containing 140 LEDs.

### Install Method

Self-Compiled

### What version of WLED?

v0.15.1+

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-08-22:
APA102 use SPI driver. Only one HW SPI is avaiable and SW SPI has issues with recent versions (i.e. not enough CPU resources to drive glitch-free output).

kylix-rd, 2025-08-22:
In version v0.14.x, it would (I presume) simulate spi on any set of pins. As stated in the bug, I was running 4 separate strips successfully. Where is this stated that only one apa102 strip is now supported? Why not leave the previous mechanisms in place and explain that it may be glitchy or run at lower frame-rates?

DedeHai, 2025-08-22:
> In version v0.14.x, it would (I presume) simulate spi on any set of pins.

A lot has changed since then. There is always the option to keep running that version. We rather provide something stable than a somewhat stable version with glitches and having many reported issues to deal with.

netmindz, 2025-08-28:
Given apa102 has much greater speed, why so few LEDs per pin?

kylix-rd, 2025-08-28:
> Given apa102 has much greater speed, why so few LEDs per pin?

It made running the wires a little simpler by breaking them up into separate channels. In hindsight, had I known this was the direction things were going, I might have done it a little differently. I've also looked over the changes and re-enabling the multiple bit-banged two-wire channels might be possible with a local patch.

willmmiles, 2025-08-29:
> If someone will confirm that WLED with several APA102 strips works without issues when under heavy load (multiple web requests while having overlapping 2D segments and running computationally demanding effect like Soap)

I wonder if the protocol is amenable to the same interrupt management logic that the WS2812 bit-banging code uses -- keep interrupts off while sending the data for one pixel, check for pending ISRs, repeat.  If so, a small driver patch would clear it right up.

sam9032, 2025-09-04:
Same problem for me.. I installed two strips on one esp32 and thought the hardware has broken. It is not easy for me to run a cable between the two strips.

blazoncek, 2025-09-05:
@sam9032 there is nothing in the code that would prevent 2 APA102 outputs. You will need to help us debug the issue by installing debug build and monitoring serial output.

beaubrewer, 2025-10-04:
I'm happy to hear I'm in good company. I had APA102s on multiple channels for cabinet lighting.

Home assistant automatically updated my device firmware. Sadly, something that has worked flawlessly for 4+ years broke with the 15.0 update.

I'd love a patch that would get me back in action.

maxroseGIT, 2025-12-10:
> [@blazoncek](https://github.com/blazoncek) i can try that.

@sam9032 did you do any debugging on using more than one APA102 strips on different pins?

DedeHai, 2026-01-15:
@willmmiles I found that all supported 2-pin LED types with WS2801 being the exception use start/stop/latch frames of zeroes and/or ones and should therefore be immune to any timing issues: they are all clock-edge driven. WS2801 uses a 500µs reset time so potentially prone to glitches.
So with the high-priority RMT driver in place, there should not be any issues with multiple bit-banged SPI outputs on APA102, LPD6803, LPD8806 and P9813. I have no hardware to test any of this though.