wled/WLED #4875: Alexa syncing issues with v0.15.1

AlexBartles, 2025-08-25:
### What happened?

When trying to connect a WLED controller/box to Alexa via the emulation built into WLED, it won't work.

### To Reproduce Bug

Do a scan for devices in the alexa app or by saying "Alexa, discover devices."

### Expected Behavior

It to sync first try or a couple of attempts in.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED v0.15.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

There is a fix I used! I found that if you unplug (as in unplug power from the devices) the ESP's power and the alexa's at the same time, it will sync if you quickly do a scan with the app! Tell me and others if this works below!

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

AlexBartles, 2025-11-21:
> Hello,
> do i have to re-plug all my echo´s?

Not entirely sure, you may only have to unplug one and it detects the lights. But I did unplug the one in each room and do it that way with the Alexa

netmindz, 2026-04-09:
> Thanks, getting the following error after downloading and extracting firmware-esp8266_2m.zip, shall I ignore this or will it brick my controller?
> 
> Firmware release name mismatch: current='ESP8266', uploaded='ESP02'. Enable 'Ignore firmware validation' to proceed anyway.

You need the regular esp8266 not the _2m

closed 2026-09-20 by github-actions[bot]

netmindz, 2026-09-22:
Resolved in https://github.com/wled/WLED/pull/5489