wled/WLED #4899: E1.31/ArtNet (DMX) group pixels

martinse1, 2025-09-01:
When using long LED-Strips with "Multi RGB" with E1.31 or ArtNet , one DMX-Universe is filled up with 170 Pixels.
So a lot of DMX-Channels are used, when in many Cases grouping of 2 or more Pixel would be fine, so save DMX-Channels.

The Topic was already discussed in Issue #2325, but not really solved.

I would suggest a "Grouping"-Parameter in Network-DMX-input-Section, where I can choose 1-n Pixel to be grouped.

For example when Choosing "4", with Multi RGB:
Pixel 1-4 get DMX-Channels 1-3 (RGB)
Pixel 5-8 get DMX-Channels 4-6 (RGB)
and so on.

That would also work with RGBW, Dimmer RGB, etc.

Regards
         Martin

axwell, 2025-11-12:
there is already a segment option for effect, maybe it would be beneficial for RGB also.

netmindz, 2025-11-12:
I believe this is already supported if you use the segment grouping value

axwell, 2025-11-12:
> I believe this is already supported if you use the segment grouping value

segment grouping value ? where is that ?

blazoncek, 2025-11-12:
Use "main segment only" in sync settings and use only one segment.