wled/WLED #4906: ESP8266 DMA method (pin 3) apparently uses all memory

willmmiles, 2025-09-03:
### What happened?

When using pin 3 for digital output on ESP8266, if the length is exactly right such that the buffer size needed for the  3-bit I2S cadence works out to a multiple of 4 bytes, then the system appears to run out of memory.

This seems to be caused by an off-by-one error in the DMA buffer translation introduced in NeoPixelBus v2.8.1.

The overrun typically squashes the "block size" of the next block of heap -- often the majority of the free RAM -- so the system reports no available memory and is unable to respond on the web interface.

### To Reproduce Bug

On an ESP8266, set a WS2812 strip of length 300 on pin 3.

### Expected Behavior

LED output should proceed and the system should respond on the web interface

### Install Method

Self-Compiled

### What version of WLED?

0.15.1, 0.16

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Anything else?

Reference discussion in #4614.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-09-03:
thanks for chasing this down.

willmmiles, 2025-09-03:
Upstream PR: https://github.com/Makuna/NeoPixelBus/pull/894