wled/WLED #4912: [Bug] WebUI become unavailable when switching wifi

MariusROBERT, 2025-09-05:
### What happened?

After coming back home from a week-end, my ambilight setup with wled and Hyperhdr no longer work, when I try to access the web interface, I got the error `net::ERR_CONTENT_LENGTH_MISMATCH`. I tried to wget the wled url but I got `Length: 44809` then `Connection closed at byte 2939`.

`http://<url>/json/state` worked and I managed to change the LED color and luminosity with wget request and the JSON API.

I tried to reset and re-install wled like 5 times with the esp-tools binary and I always got the same bug after switching from AP to my home wifi.

I already had this bug a few month earlier and it fixed itself in around 2-3 weeks, I still don't know how.

### To Reproduce Bug

Fresh install wled and connect it to my wifi

### Expected Behavior

Led should turn on and wled webUI should still be available.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.1 (build 2507300)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

willmmiles, 2025-09-05:
What sort of wifi access point do you have?   Curiously I observed something very similar to this last week with my bench 8266 - it worked fine in AP mode, but could barely get any packets through when connected to the wifi.  Changing the WLED build and resetting the device configuration didn't help.   However, the problem vanished when I rebooted my AP -- the device started working properly immediately after it reconnected.  (My test network is currently on a MikroTik device.)

blazoncek, 2025-09-06:
@willmmiles I can corroborate that.

Often misconfigured WiFi network will cause issues with ESP devices. Two most important things that need to be turned off are Fast roaming and BSS transition.
However, sometimes other settings may interfere too. But that is not all to it.

Just a few days ago we had a power surge which left 1/3 of my IoT devices unable to communicate. I had to deauthorize them AND reboot AP they connected to even though my network is powered through UPS.

MariusROBERT, 2025-09-14:
Allowing access to only one wifi node (which is around 20cm away from it) to my esp didn't fixed the problem but switching to the `compat` build does