wled/WLED #4920: NOWIFISLEEP build flag not honored

rhkean, 2025-09-10:
### What happened?

I noticed that the NOWIFISLEEP build flag was not being honored and always set true for esp32's and false for esp8266's.

I'm committing a PR for it

### To Reproduce Bug

* add `DEBUG_PRINTF_P(PSTR("calling WiFi.setSleep(%d);"), !noWifiSleep);` just before the call in the `initConnection` function on `wled.cpp`
* see that `setSleep` is always called with `0`, regard of the build flag

### Expected Behavior

-D NOWIFISLEEP=0 enables WiFi sleep

### Install Method

Self-Compiled

### What version of WLED?

nightly

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-09-10:
there is no buildflag called NOWIFISLEEP.

rhkean, 2025-09-10:
discussion on PR opted to not include the build flag

closed 2025-09-10 by rhkean