wled/WLED #4958: Reboot instead of restricting LED-type change with multiple buses

DedeHai, 2025-09-24:
When adding more than one bus in LED settings, editing the type is only possible for the last bus.
This is to ensure proper hardware initialisation. 
However, this is confusing to users and should be handled differently. Instead of restricting it in the settings, a variable should be added to the LED settings page that keeps track of the problematic changes i.e. the bus type and if necessary, display a warning and issue a reboot.
@willmmiles do you know which settings or combinations are critical in this regard?

willmmiles, 2025-09-28:
Honestly I don't know why the restriction was originally implemented, beyond noting that I'm aware that construction order can matter so it isn't possible in the general case to modify bus settings in place without (at the very least) destroying them all and reconstructing.  And reallocating bus resources can in turn be problematic if the system RAM is fragmented as it can be tough to make large allocations, leading to failed Bus objects and probable crashes as those "unhappy" states are less well tested.  A good example is the trouble with long strips on ESP8266 -- it can work OK after a reboot, but gets dicey if the settings are changed mid-run as the new LED buffer might fragment available RAM such that segment/pixel buffers are harder to allocate.

Similarly, driver cleanup is probably just less well tested.  I don't know of any specific issues offhand, but then the UI change was a response to problems that came from before I started working on the project.  There might be some tracks in the issue board or forums.

DedeHai, 2025-10-05:
@blazoncek can you explain on why this restriction was needed?

blazoncek, 2025-10-05:
To correctly account available buses.

A workaround would be to add drag & drop bus reordering funcionality of buses in settings. That would allow user to re-arrange buses without needing to delete them.

Unfortunately that was beyond my HTML/JS expertise.

DedeHai, 2025-10-22:
https://github.com/wled/WLED/pull/5014  properly fixes this.