wled/WLED #4959: Better error logging / info for users

DedeHai, 2025-09-24:
As recently discussed, current error display using toast-messages is not very verbose and do not really tell the user what is going on. 
An error log that users can access with more descriptive messages and that holds the last 5 errors or so would be much better. 
My suggestion is this:
- keep the toast messages as an immediate alert
- in addition, turn the info symbol in the UI into a red exclamation mark until the user clicks on it
- add a text field at the bottom of the info panel (below the buttons) that is hidden by default and shows the last errors 
- add more descriptive strings to be displayed in that text field
- add more error types to better distinguish different issues (no ram for bus, no ram for global buffer, no ram for segment, no ram for tansitions, pin conflicts, ...)

DedeHai, 2025-09-27:
what the copilot came up with can be used as a base. I updated the styling a bit (it was awful) and now I think this could work:

[image]

this what the error display could look like: animation takes only like 5 lines of code, may be too agressive and annoying though

[image]

what do you think @willmmiles ?

I would suggest to first do a PR as a base, adding the UI elements and then build upon that.

DedeHai, 2025-09-27:
> The icon blinking is good, as long as it stops blinking when you click it.

I found the blinking can be limited, like 10 times and then go to static. it already goes back to normal when you click on it.

 > On the specific UI: I'd lean towards putting the errors at the top,

sure, easy to move if that does not work out.

> * Getting rid of the bottom-based toast popups that cover control elements.  

I found them not an issue but I use it on PC mostly. Can imagine they are annoying on a mobile. will remove.

> * Better concept of state/persistence with errors -- both a real log (with timestamps!) and a way to indicate which ones are still "live".

they were with timestamps from copilot as instructed, I did not find it helpful to know the exact time but have no strong opinion.

> * Replacing the codes with strings, so we can include context in the message.  (Eg. which segment failed, or what bus is having a problem, and so on.)  Bandwidth is cheap, and we have to store the error strings anyways (be it in the UI or in the code).

if its beneficial in some way, sure. strings on web UI are gzipped, so cheaper flash wise.

willmmiles, 2025-09-27:
> > * Better concept of state/persistence with errors -- both a real log (with timestamps!) and a way to indicate which ones are still "live".
> 
> they were with timestamps from copilot as instructed, I did not find it helpful to know the exact time but have no strong opinion. I did limit the number of errors to 10 currently, so only latest 10 are kept until discarded. since they are stored on the UI, no log is kept if for example controlling from HA or buttons.

Right, that's the issue I'd like to improve on: the current status and history needs to be kept on the device.  Clients should be stateless - refreshing the page, or having multiple clients connected, shouldn't make messages disappear.  "I saw an error, then refreshed and it was gone" is IMO not helpful for figuring out why one of my segments isn't working.

Re timestamps: as much as I truly loathe the way GitHub hides the real timestamps and says "5 minutes ago", even relative timestamps can be very helpful for identifying which is the broken preset.

> if its beneficial in some way, sure. strings on web UI are gzipped, so cheaper flash wise.

The first use case that comes to mind is saying which segment failed to allocate with what effect -- that way if I'm pushing the limits of my device, I can see which things pushed it over the edge.   Or also, as above, identifying the preset that called for the unsupportable state.