wled/WLED #4963: custom color palette box

adamjensen61, 2025-09-27:
### What happened?

Im in v16.0 release 27/2025, whenever i click on the box to change color the box does not appear

### To Reproduce Bug

the color palette box will not show

### Expected Behavior

there is no box to change to different color

### Install Method

Binary from WLED.me

### What version of WLED?

16.0 release 27/2025

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-09-27:
not really clear to me what you mean by "color palette box" but if you use the latest code, color changing and custom palettes work.

adamjensen61, 2025-09-27:
> not really clear to me what you mean by "color palette box" but if you use the latest code, color changing and custom palettes work.

i meant the custom color palette, like when you goto the custom color palette for making your own color and then you click on the little box to skim through colors, that box does not pop up for me

DedeHai, 2025-09-27:
thanks for the clarification. bug confirmed, will fix it.

commit e23751bd (2025-09-27): Bugfix in custom palette color picker, fixes #4963
closed 2025-09-27 by DedeHai