wled/WLED #4965: Add a weather display for the matrix.

Sergeywhiteair, 2025-09-28:
Please add a new effect for the matrix that will display the weather with icons: clouds, rain, sun, snow. You can make the icons animated. Next to the weather conditions icon, display the air temperature in C and F (with the option to select either C or F in the settings)

DedeHai, 2025-09-28:
this would need to be a usermod. There are other implementations already:
https://mm.kno.wled.ge/usermods/Weather/
https://github.com/wled/WLED/pull/4883

Sergeywhiteair, 2025-09-28:
> Для этого нужен пользовательский мод. Есть и другие реализации: [https://mm.kno.wled.ge/usermods/Weather/ ](https://mm.kno.wled.ge/usermods/Weather/) [#4883](https://github.com/wled/WLED/pull/4883)

Спасибо за ответ!

closed 2025-09-28 by Sergeywhiteair